DreamSourceLab/DSView #815: Decode result list - source reset after restart of program

Tobey16, 2024-11-07:
After each restart of the program, decoding results will always show the results of the first decoder that is in the decoder list, regardless of what has been setup before.

dreamsource-tai, 2024-11-11:
@Tobey16 
Sorry, I don't quite understand what you mean.

Tobey16, 2024-11-11:
The decoder is always beeing reset to the first decoder (Timing) in the list after restarting DSView, when before another decoder was set in the decoding results list.
[image]

dreamsource-tai, 2024-11-12:
@Tobey16 
The decoder list above, the ones added first are listed first. The decoder list in the decoding result may not be arranged in the same order as the list above. I hope my reply can be helpful to you.

Tobey16, 2024-11-12:
No, it does not help. The issue is not resolved, as i have to set it over and over again after restarting the program.

BTW: Have you seen my other issues? (#812 - #814)

dreamsource-tai, 2024-11-12:
@Tobey16 
Have you tried the v1.3.2?
I test this issue in my system, but it is ok.

dreamsource-tai, 2024-11-12:
@Tobey16 
Restarting the program will definitely reset the list, but it will only be a recollection and the list will not be affected. What I mean is the selected decoder in the decoding result.

Tobey16, 2024-11-12:
I am usind 1.3.2. I have to select UART on every start of the program, because Timing is set on every start of the program.