DreamSourceLab/DSView #821: DSLogic U3Pro32 is always seen as a USB2.0 device in DSView

croz-tech, 2024-11-24:
Per the title my U3Pro32 is always seen as a USB2.0 device, despite being connected to either a USB-C 3.0 port or a USB-A 3.1 port.  I'm using DSView v1.3.2 on Windows 10.  Any thoughts?  Might the OS be picking up the wrong driver?  This lists as `USB-based DSL Instrument v2` with Driver Version `1.0.0.0`  Note, I've NEVER got this to recognize as a USB3 device either on PC or Mac.
Any suggestions are appreciated, Steve

dreamsource-tai, 2024-11-25:
@croz-tech 
Are you using the original data cable? Have you used a HUB?

croz-tech, 2024-11-25:
Not at the moment. Can you post a picture of the original cable and I will dig it out/try that one.

dreamsource-tai, 2024-11-26:
@croz-tech 
Are you sure your USB cable and PC port are both 3.0?  and, try to switch the side of cable head.

croz-tech, 2024-11-27:
@dreamsource-tai  I have dug out what I believe to be the original USB-A to C cable and tried that.  I tried it with the connector rotated and tried three different USB-C to C cables in all orientations.  In every case I got the same message about the device being seen as USB 2.0

dreamsource-tai, 2024-11-28:
@croz-tech 
Please contact support@dreamsourcelab.com Obtain after-sales support.

dannydulai, 2025-04-10:
I saw the same thing on MacOS -- MacBook Pro 16, Nov 2023. There are no USB 2.0 ports on this machine. I can not use the cable the U3Pro32 came with because it is USB-A to USB-C and this machine has no USB-A ports.

However, I tried with a USB-C SS10 cable I had lying around and it no longer complained.