DreamSourceLab/DSView #823: Capture & Replay

vbooka1, 2024-12-01:
Hello, is there any chance to see a "replay" feature in the future? i.e. to use the DSLogic analyzer as a pattern generator, to be able to capture some signal stream and then "replay" it, for example to emulate some hardware.

m72m72, 2024-12-01:
Hello!

Most likely not because of the hardware architecture. All inputs have high input resistance. And for the output, on the contrary, low resistance is required.
For playback, I recommend paying attention to the products of Digilent:
- Digital Discovery (has a 2 Gigabit buffer)
- Analogue Discovery 2/3 (can be an oscilloscope)
But the price...
You can download the WaveForms application and study it in demo mode.

Best regards.

0wwafa, 2025-01-31:
@m72m72 that's not true. I have an fx2lp device and it works perfectly both as input both as output.

m72m72, 2025-01-31:
Ok!  Do you know schematic of the DS Logic inputs? Are the FPGA pins connected directly to the connector? How will you pass signal through input resistor 250K?

0wwafa, 2025-02-13:
@m72m72 my device is a "logic analyzer"  from the 8 i/o pins outside the signal goes through a pull down resistor (100 ohm/100k ohm) and the directly to the chip port B pins.
I can do I/O with it.