DreamSourceLab/DSView #825: Arch Linux  下 dsview-git 包编译错误

taotieren, 2024-12-14:
```bash
[ 93%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/output/gnuplot.c.o
[ 93%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/output/srzip.c.o
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/demo/demo.c:75:6: warning: implicit conversion from ‘enum DEMO_LOGIC_CHANNEL_ID’ to ‘enum DEMO_CHANNEL_ID’ [-Wenum-conversion]
75 |     {DEMO_LOGIC125x16,  LOGIC,  SR_CHANNEL_LOGIC,  16, 1, SR_MHZ(1), SR_Mn(1),
|      ^~~~~~~~~~~~~~~~
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/demo/demo.c:77:6: warning: implicit conversion from ‘enum DEMO_LOGIC_CHANNEL_ID’ to ‘enum DEMO_CHANNEL_ID’ [-Wenum-conversion]
77 |     {DEMO_LOGIC250x12,  LOGIC,  SR_CHANNEL_LOGIC,  12, 1, SR_MHZ(1), SR_Mn(1),
|      ^~~~~~~~~~~~~~~~
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/demo/demo.c:79:6: warning: implicit conversion from ‘enum DEMO_LOGIC_CHANNEL_ID’ to ‘enum DEMO_CHANNEL_ID’ [-Wenum-conversion]
79 |     {DEMO_LOGIC500x6,  LOGIC,  SR_CHANNEL_LOGIC,  6, 1, SR_MHZ(1), SR_Mn(1),
|      ^~~~~~~~~~~~~~~
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/demo/demo.c:81:6: warning: implicit conversion from ‘enum DEMO_LOGIC_CHANNEL_ID’ to ‘enum DEMO_CHANNEL_ID’ [-Wenum-conversion]
81 |     {DEMO_LOGIC1000x3,  LOGIC,  SR_CHANNEL_LOGIC,  3, 1, SR_MHZ(1), SR_Mn(1),
|      ^~~~~~~~~~~~~~~~
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/demo/demo.c: In function ‘config_set’:
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/demo/demo.c:1138:41: warning: implicit conversion from ‘enum DEMO_CHANNEL_ID’ to ‘enum DEMO_LOGIC_CHANNEL_ID’ [-Wenum-conversion]
1138 |                     vdev->logic_ch_mode = (enum DEMO_CHANNEL_ID)nv;
|                                         ^
[ 93%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/output/vcd.c.o
[ 93%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/hardware/DSL/dslogic.c.o
[ 94%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/hardware/common/usb.c.o
[ 94%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/hardware/common/ezusb.c.o
/build/dsview-git/src/DSView/libsigrok4DSL/output/csv.c: In function ‘init’:
/build/dsview-git/src/DSView/libsigrok4DSL/output/csv.c:73:9: warning: variable ‘ch_num’ set but not used [-Wunused-but-set-variable]
73 |     int ch_num;
|         ^~~~~~
/build/dsview-git/src/DSView/libsigrok4DSL/output/csv.c: In function ‘gen_header’:
/build/dsview-git/src/DSView/libsigrok4DSL/output/csv.c:150:16: warning: variable ‘t’ set but not used [-Wunused-but-set-variable]
150 |         time_t t;
|                ^
/build/dsview-git/src/DSView/libsigrok4DSL/output/csv.c: In function ‘receive’:
/build/dsview-git/src/DSView/libsigrok4DSL/output/csv.c:301:46: error: format not a string literal and no format arguments [-Werror=format-security]
301 |                 g_string_append_printf(*out, tmp_buffer);
|                                              ^~~~~~~~~~
/build/dsview-git/src/DSView/libsigrok4DSL/output/csv.c:368:26: warning: comparison of integer expressions of different signedness: ‘uint64_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
368 |            for (j = 0; j < ch_num; j++) {
|                          ^
[ 94%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/trigger.c.o
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/DSL/dslogic.c: In function ‘scan’:
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/DSL/dslogic.c:292:5: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
292 |     else
|     ^~~~
/build/dsview-git/src/DSView/libsigrok4DSL/hardware/DSL/dslogic.c:295:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
295 |         conn = NULL;
|         ^~~~
cc1: some warnings being treated as errors
make[2]: *** [CMakeFiles/DSView.dir/build.make:3703: CMakeFiles/DSView.dir/libsigrok4DSL/output/csv.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 95%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/dsdevice.c.o
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/DSView.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
==> ERROR: A failure occurred in build().
Aborting...
==> ERROR: Build failed, check /var/lib/archbuild/extra-x86_64/lilac-0/build
```

dreamsource-tai, 2024-12-16:
@taotieren 
谢谢你的反馈。这个编译出错可能跟每人的库版本和编译环境有关。我这边在archlinux下试过，能编译的。

taotieren, 2024-12-16:
我使用的是 dsview-git 包的 pkgbuild  编译环境是 devtools 包里面的 extra-build 进行编译测试的。我这边的 打包环境就是devtools 里面的提供的“干净”环境