DreamSourceLab/DSView #832: Protocol decode problem in DSView v1.3.2

kintekobo, 2024-12-30:
Running DSView v1.3.2 with DSLogic Plus in MacOs Sonoma 14.6.1 and MacMini Apple M2 Pro. 

When capturing an SPI transmission from an RP2040 I noticed that although the trigger was firing at the start of the first clock burst as requested, the data was decoding one clock pulse later, resulting in a garbled decode. I’m pretty sure that the same settings had decoded correctly previously so I am confused as to what is wrong. I have tried starting the trigger point at the fall of the CS but the decode is exactly the same, as is using simple trigger.

I have attached the DSL file (named as .TXT), the Log file and two screen caps showing the first and second stage of the trigger.

Hoping that you can say why this is happening.

Chris

[log.txt](https://github.com/user-attachments/files/18277184/log.txt)
[image]
[image]
[DSLogic PLus-la-241230-190432.txt](https://github.com/user-attachments/files/18277200/DSLogic.PLus-la-241230-190432.txt)

DreamSourceLab, 2024-12-31:
SPI has 4 modes, as shown in the figure below, which specify different edge alignment methods. 
[image]

If you want to decode on the rising edge, simply set both CPHA and CPOL to 0. As shown in the figure below.
[image]

kintekobo, 2024-12-31:
Thanks for the reply, however the settings you suggest are the ones that I am using. The reason that I think that it may be a bug is that shutting down the program and restarting it with the exact same settings will often result in the decode being correct.