DreamSourceLab/DSView #838: 功能需求：支持通过终端或网络控制 DSView

Halley826, 2025-01-05:
**需求描述：**  
我们希望在自动化测试中使用逻辑分析仪，但目前 DSView 不支持通过终端或网络命令进行操作。我们请求添加此功能，以便实现自动化。  

---

### 功能详情：自动捕获逻辑分析仪数据  

**问题描述：**  
我们的嵌入式设备需要进行大量 QA 测试，但测试结束后缺乏详细的 GPIO 日志信息，导致一些问题无法深入分析。  

**解决方案：**  
希望在测试用例开始时能够自动捕获 GPIO 信号，在测试用例结束后自动停止并保存 GPIO 日志。虽然可以手动完成这些操作，但由于测试用例数量庞大且测试时间较长，手动保存难以满足需求。  

**目前遇到的问题：**  
当前 DSView（v1.3.0 版本）不支持自动化的捕获、停止和保存 GPIO 日志。缺少此功能，使得测试过程的自动化受到限制。  

---

**请求的改进：**  
1. 增加通过终端命令或网络接口控制 DSView 的功能。

dreamsource-tai, 2025-01-06:
@Halley826 
我们的代码是开源的，想实现你这样的需求不难。可以自己改成你们想要的功能。

Halley826, 2025-01-10:
@dreamsource-tai 好的，谢谢，我自己做了该功能。
我使用 stream 模式（plus)，对电脑的内存的配置有要求吗？

dreamsource-tai, 2025-01-14:
@Halley826 
stream模式就是采集到信息马上传到上位机，buffer就是采集数据到硬件，达到采集量后再一次性传到上位机。stream的存储深度高一些。

HelloMrWan, 2025-03-06:
> [@dreamsource-tai](https://github.com/dreamsource-tai) 好的，谢谢，我自己做了该功能。 我使用 stream 模式（plus)，对电脑的内存的配置有要求吗？

您好，我们也有类似的需求，请问是否可以提供一下您的实现这个功能的思路呢？您是在linux系统下修改源码的吗？我们用的windows系统，但在windows下编译太麻烦了