DreamSourceLab/DSView #839: DSView-1.3.2 build errors on Ubuntu 24.04 with Qt6

xealits, 2025-01-05:
I just downloaded the recent version of [DSView from the website](https://www.dreamsourcelab.com/download/), and following INSTALL and building it with Qt6, the `make` runs into these errors:

```
.../DSView-1.3.2/DSView/main.cpp:32:
/.../DSView-1.3.2/DSView/pv/mainframe.h:117:10: error: ‘bool pv::MainFrame::nativeEvent(const QByteArray&, void*, long int*)’ marked ‘override’, but does not override
  117 |     bool nativeEvent(const QByteArray &eventType, void *message, long *result) override;
      |          ^~~~~~~~~~~

.../DSView-1.3.2/DSView/pv/mainwindow.cpp:102:
/.../DSView-1.3.2/DSView/pv/mainframe.h:117:10: error: ‘bool pv::MainFrame::nativeEvent(const QByteArray&, void*, long int*)’ marked ‘override’, but does not override
  117 |     bool nativeEvent(const QByteArray &eventType, void *message, long *result) override;
      |          ^~~~~~~~~~~

/.../DSView-1.3.2/DSView/pv/mainframe.cpp: In member function ‘bool pv::MainFrame::nativeEvent(const QByteArray&, void*, long int*)’:
/.../DSView-1.3.2/DSView/pv/mainframe.cpp:1135:53: error: invalid conversion from ‘long int*’ to ‘qintptr*’ {aka ‘long long int*’} [-fpermissive]
 1135 |     return QWidget::nativeEvent(eventType, message, result);
      |                                                     ^~~~~~
      |                                                     |
      |                                                     long int*

/.../DSView-1.3.2/DSView/pv/dock/keywordlineedit.cpp: In member function ‘virtual void KeyLineEdit::wheelEvent(QWheelEvent*)’:
/.../DSView-1.3.2/DSView/pv/dock/keywordlineedit.cpp:123:24: error: ‘class QWheelEvent’ has no member named ‘delta’
  123 |             if (event->delta() > 0){
      |                        ^~~~~
```

To build it, I installed Qt6 (apt packages `qt6-base-dev` and `qt6-tools-dev`, there is no `qt6-default` and neither `qt5-default` on Ubuntu 24.04). I tried to correct the `bool pv::MainFrame::nativeEvent` at `mainframe.h:117`: comment out `override` and change the result pointer parameter to `long long*`. But I can't do anything with "`QWheelEvent` has no member `delta`".

Is DSView-1.3.2 expected to work with Qt6 on Ubuntu, or should I build it with Qt5 somehow?

dreamsource-tai, 2025-01-06:
@xealits 
DSView在qt6编译出来的程序，形为有些不完善。请用qt5编译。

dreamsource-tai, 2025-01-06:
@xealits 
The program compiled by DSView in Qt6 has some incorrect shapes. Please compile using qt5.

dreamsource-tai, 2025-01-06:
@xealits 
After compiling DSView with QT6, there is an error in dragging the window size. Please compile it under QT5.

xealits, 2025-01-06:
@dreamsource-tai I run it in a tiling window manager, [i3](https://i3wm.org/), and resizing seems to work fine, although I did not test it on a real device.

I.e. when I open a different window, DSView adjusts, horizontal side-scroller shows up. There is no vertical scroller, DSView just makes the font of the lines smaller. And when you resize the DSView window in i3, with `Command+R`, it also works. With Demo Device, I can set the Mode to loop and get random signals run on Start.

Just to confirm, `ldd DSView` does show that it links to Qt6:
```
libQt6Widgets.so.6
libQt6Gui.so.6
libQt6Core.so.6
libQt6DBus.so.6
```
Although, it turns out I do have Qt5 libs in `/lib/x86_64-linux-gnu/`.

closed 2025-01-13 by xealits