DreamSourceLab/DSView #840: Decoder list empty

urobasa, 2025-01-10:
Ubuntu 24.04
git clone
build
install
run
signal capture done
decoder search opened
empty list show
[image]
`DSView: Set default device.
sr: lib_main: Release current actived device. name:"DSLogic PLus", handle:0x5a11fee65740
sr: dsl: DSLogic: Closing device 0 on 1.4 interface 0.
sr: lib_main: Start activating device.
sr: lib_main: Close the previous device, name:"DSLogic PLus"
sr: lib_main: Activating virtual device name: "DSLogic PLus-la-250110-024627.dsl".
sr: lib_main: To open device, name:"DSLogic PLus-la-250110-024627.dsl"
sr: virtual-session: The 'header' file format version:3
sr: lib_main: Activating device end.
DSView: Switch to file "DSLogic PLus-la-250110-024627.dsl" done.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
DSView: SigSession::reload(), clear signals
DSView: Update rate list.
DSView: Update sample count list.
DSView: StoreSession::load_decoders(), json object array is empty.`

dreamsource-tai, 2025-01-10:
@urobasa 
Please see the log, and check the  directory of decoders.
Do you run DSView from the compile output directory, or the "/user/bin"?

urobasa, 2025-01-12:
on fresh install ubuntu work.
create deb pack with checkinstall and install to my system - work.

urobasa, 2025-01-12:
local linux system invisible error?

closed 2025-01-12 by urobasa