DreamSourceLab/DSView #847: CMake error on Ubuntu 22.04.5 LTS - Could NOT find Python3

Solovok, 2025-01-29:
Hi!

I'm try to compile DSView 1.3.2 on my system and get this error:

```
$ cmake .
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for one of the modules 'glib-2.0'
----- glib-2.0:
-- 	 includes:/usr/include/glib-2.0/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- 	 libraries:/usr/lib/x86_64-linux-gnu/libglib-2.0.*
-- Could NOT find Python3 (missing: Python3_LIBRARY Python3_INCLUDE_DIR Development) 
CMake Error at CMakeLists.txt:103 (message):
  Please install lib python3!

-- Configuring incomplete, errors occurred!
See also "/home/costa/Загрузки/DSView-1.2.0/CMakeFiles/CMakeOutput.log".
```

```
$ python --version
Python 3.11.0rc1
$ python3 --version
Python 3.10.12
```

Installing the requirements is OK:
```
$ sudo apt install git gcc g++ make cmake qtbase5-dev libglib2.0-dev zlib1g-dev libusb-1.0-0-dev libboost-dev libfftw3-dev python3-dev libudev-dev pkg-config
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово         
Уже установлен пакет g++ самой новой версии (4:11.2.0-1ubuntu1).
Уже установлен пакет gcc самой новой версии (4:11.2.0-1ubuntu1).
Уже установлен пакет libboost-dev самой новой версии (1.74.0.3ubuntu7).
Уже установлен пакет libfftw3-dev самой новой версии (3.3.8-2ubuntu8).
Уже установлен пакет make самой новой версии (4.3-4.1build1).
Уже установлен пакет pkg-config самой новой версии (0.29.2-1ubuntu3).
Уже установлен пакет cmake самой новой версии (3.22.1-1ubuntu1.22.04.2).
Уже установлен пакет git самой новой версии (1:2.34.1-1ubuntu1.12).
Уже установлен пакет libglib2.0-dev самой новой версии (2.72.4-0ubuntu2.4).
Уже установлен пакет libudev-dev самой новой версии (249.11-0ubuntu3.12).
Уже установлен пакет libusb-1.0-0-dev самой новой версии (2:1.0.25-1ubuntu2).
Уже установлен пакет python3-dev самой новой версии (3.10.6-1~22.04.1).
Уже установлен пакет zlib1g-dev самой новой версии (1:1.2.11.dfsg-2ubuntu9.2).
Уже установлен пакет qtbase5-dev самой новой версии (5.15.3+dfsg-2ubuntu0.2).
```
How I can fix it?

dreamsource-tai, 2025-02-05:
Please try to install the packages by apt.

Solovok, 2025-02-05:
> Please try to install the packages by apt.

What packages should I install? Are there any deb packages for DSView?

dreamsource-tai, 2025-02-06:
@Solovok 
There's no packages on ubuntu, and compile the project so easy on ubuntu.
Please see the file INSTALL, it shows the way to compile DSView.
I guess your library of python couldn't be found, maybe because there's no. cmake file available.
So try the apt.