DreamSourceLab/DSView #848: Cannot threshold levels <0.6V

stefanct, 2025-02-04:
Hi,
I need to debug USB signals and thus need to set a threshold level of ca. 0.2V. For some reason the `bind_double` function makes this impossible and forces all values <0.6 to 5.0 (in version 2e9e2c8e726df4ef5687d39b83d4f797cc44b574 on Debian 12). If I change the `bind_double` call

https://github.com/DreamSourceLab/DSView/blob/2e9e2c8e726df4ef5687d39b83d4f797cc44b574/DSView/pv/prop/binding/deviceoptions.cpp#L110

to the following, I get better behavior:

            bind_double(name, label, key, "V", pair<double, double>(0.0, 5.0), 2, 0.05);

With that I think the threshold is set to a/the lower value. Unfortunately, I think my setup has too much noise at the moment to get a proper signal but at least I have a chance now ;)

dreamsource-tai, 2025-02-05:
@stefanct 

[image]

stefanct, 2025-02-05:
That's great that it works for you but does not explain why it doesn't in my case ;)

I played around a bit more and my current hypothesis is that it is (again) related to my locale and its use of `,` as decimal separator. The reason for my suspicion is that I cannot enter a `.` when editing the text field manually - I have to use a `,` and it gets converted to `.` afterwards (i.e., when the focus leaves the field). And then the smaller thresholds actually work! However, when I use the arrow buttons to adjust it, it directly uses `.` but when the focus leaves the value gets converted to the 10x value (e.g., 0.4 becomes 4.0).

I have now also confirmed this by starting dsview via `LC_NUMERIC="en_US.UTF-8" .../DSView` and with that, I cannot reproduce the problem. You probably can reproduce my problem when using a locale that uses comma as decimal separator.

I have no idea why it does work with 0.05V steps though... QT is a mystery sometimes ;)