DreamSourceLab/DSView #849: Cannot use DSView on second monitor

rob13572468, 2025-02-05:
On Mac OS (15.3) When I try to drag the DSview app to my second monitor it snaps back to the main monitor.

dreamsource-tai, 2025-02-06:
@rob13572468 
Please arrange your two monitors horizontally instead of vertically.

jnewcomb, 2025-02-21:
I've got three monitors side by side in portrait mode (good for columns of code)..  but.. I can't extend the window beyond a single monitor's width- it's unsuitable for looking at long traces.
Im using DSLogic Plus on Windows 11 - Is there an older version that did not have this problem?
Thanks..

bigtreetech, 2026-01-19:
> [@rob13572468](https://github.com/rob13572468) Please arrange your two monitors horizontally instead of vertically.

Is there a fix for this yet where we can stack monitors vertically? Is V1.3.3 RC available for macos yet?