DreamSourceLab/DSView #850: export between cursors compressed

JoeBut, 2025-02-06:
Exporting between cursors should export data between the cursors.
What it actually does is:
Starts the export at cursor 1.
Then excludes data between cursor 1 and 2 and some more.
Then exports data after cursor 2. 
Version: 1.3.2 x64

dreamsource-tai, 2025-02-07:
@JoeBut 
Try the new version:
https://dreamsourcelab.cn/software/DSView_v1.3.3-rc3_x64_setup.exe

JoeBut, 2025-02-08:
v1.3.3-rc3 x64: 

OK: with a short capture (10 MB dsl file, 1 second, 4 ch, 500 MHz) export between cursors (compressed) works correctly.

**Bug**: with a _very long_ capture (**339 MB** dsl file, 1.76 min, 15 ch, 100 MHz) export between cursors (compressed) does not work correctly. It starts the csv at the correct timestamp, but continues exporting data beyond cursor 2. 

Note: after 2 minutes of exporting, I cancelled the export and the csv file was found to be 2 GB in size and the end timestamp was at about 2 seconds when cursor 2 should have halted at 144 ms.

_The export should stop when cursor 2 is reached._