DreamSourceLab/DSView #854: Ubuntu 24.04 Cannot move Main Window

codepox, 2025-02-10:
Hi,
I am on Ubuntu 24.04 and I built from the master branch using QT5. 
DSView launch and seems to work OK, but the main window does not move, it is not possible to move the App on the display.
Thanks,

dreamsource-tai, 2025-02-11:
@codepox 
Are you compiled the project with qt5?

codepox, 2025-02-11:
Yes, I compiled with qt5

dreamsource-tai, 2025-02-11:
@codepox 
Could you upload the log of running DSView?

codepox, 2025-02-11:
QSocketNotifier: Can only be used with threads started with QThread
DSView: ----------------- version: 1.3.2-----------------
DSView: Qt:5.15.13
DSView: 2025-02-11 18:47:30
DSView: GetAppDataDir:"/usr/local/share/DSView"
DSView: GetFirmwareDir:"/usr/local/share/DSView/res"
DSView: GetUserDataDir:"/home/gme/.local/share/DreamSourceLab/DSView"
DSView: GetDecodeScriptDir:"/usr/local/share/libsigrokdecode4DSL/decoders"
sr: lib_main: Init libsigrok.
sr: lib_main: Scan all connected hardware device.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found new DSLogic device count: 0
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
sr: backend: Register hotplug callback.
sr: lib_main: Hotplug thread start!
/usr/local/share/libsigrokdecode4DSL/decoders/ds1307/pd.py:51: SyntaxWarning: invalid escape sequence '\/'
  l += [('bit_' + re.sub('\/| ', '_', b).lower(), b + ' bit') for b in bits]
/usr/local/share/libsigrokdecode4DSL/decoders/ds1307/pd.py:55: SyntaxWarning: invalid escape sequence '\/'
  ['BIT_' + re.sub('\/| ', '_', b).upper() for b in bits] + \
/usr/local/share/libsigrokdecode4DSL/decoders/arm_etmv3/pd.py:215: SyntaxWarning: invalid escape sequence '\s'
  instpat = re.compile('\s*([0-9a-fA-F]+):\t+([0-9a-fA-F ]+)\t+([a-zA-Z][^;]+)\s*;?.*')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_etmv3/pd.py:216: SyntaxWarning: invalid escape sequence '\.'
  branchpat = re.compile('(b|bl|b..|bl..|cbnz|cbz)(?:\.[wn])?\s+(?:r[0-9]+,\s*)?([0-9a-fA-F]+)')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_etmv3/pd.py:217: SyntaxWarning: invalid escape sequence '\s'
  filepat = re.compile('[^\s]+[/\\\\]([a-zA-Z0-9._-]+:[0-9]+)(?:\s.*)?')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_etmv3/pd.py:218: SyntaxWarning: invalid escape sequence '\s'
  funcpat = re.compile('[0-9a-fA-F]+\s*<([^>]+)>:.*')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_itm/pd.py:116: SyntaxWarning: invalid escape sequence '\s'
  instpat = re.compile('\s*([0-9a-fA-F]+):\t+([0-9a-fA-F ]+)\t+([a-zA-Z][^;]+)\s*;?.*')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_itm/pd.py:117: SyntaxWarning: invalid escape sequence '\s'
  filepat = re.compile('[^\s]+[/\\\\]([a-zA-Z0-9._-]+:[0-9]+)(?:\s.*)?')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_itm/pd.py:118: SyntaxWarning: invalid escape sequence '\s'
  funcpat = re.compile('[0-9a-fA-F]+\s*<([^>]+)>:.*')
DSView: Set default device.
sr: lib_main: Start activating device.
sr: lib_main: Activating device name: "Demo Device".
sr: lib_main: To open device, name:"Demo Device"
sr: lib_main: Activating device end.
DSView: Switch to device "Demo Device" done.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
DSView: Read region info, x:81, y:254, w:2258, h:898, isMax:0
DSView: Last display:DP-2
DSView: Screen name:DP-1, region, left:2560, top:0, width:3072, height:1728
DSView: Screen name:DP-2, region, left:0, top:0, width:2560, height:1440
DSView: Restore, screen:DP-2, x:0, y:0, w:2560, h:1440
DSView: Normal region, x:81, y:254, w:2258, h:898
DSView: Init form, x:81, y:254, w:2259, h:899, k:2

dreamsource-tai, 2025-02-12:
@codepox 
Is your window maximized? When the window is maximized, it cannot be moved. It can be done under Windows.

codepox, 2025-02-12:
No the window is not maximize and I cannot move it.
Can you reproduce the problem on Ubuntu 24.04?

dreamsource-tai, 2025-02-13:
@codepox 
My system version is also 24.04, and DSView works well. My QT version is the same as yours.

dreamsource-tai, 2025-02-13:
@codepox 
Are you using a mouse or a touchpad? There are several ways to try:
1. Set the resolution to xxx * 1080;
2. Downgrade the software version to v1.2.1 (run the command "git tag" to view);
3. If there are two monitors, arrange them horizontally;

codepox, 2025-02-17:
I changed the resolution to xxx * 1080, tried one and two monitor and reverted to v1.2.1 without success.

I saw the following error, could that help:
qt.qpa.wayland: setGrabPopup called with a parent, QtWaylandClient::QWaylandXdgSurface(0x62bc709c6b40) which does not match the current topmost grabbing popup, QtWaylandClient::QWaylandXdgSurface(0x62bc7075f290) According to the xdg-shell protocol, this is not allowed. The wayland QPA plugin is currently handling it by setting the parent to the topmost grabbing popup. Note, however, that this may cause positioning errors and popups closing unxpectedly because xdg-shell mandate that child popups close before parents

codepox, 2025-02-17:
I tried to cleanup everything and reinstall everything, I removed: 
~/.config/Dreamsourcelab
~/.local/share/Dreamsourcelab
/usr/local/share/DSView
/usr/local/share/libsigrokdcode4DSL

But DSView still remembers some of my settings, are there more to clean to have a pristine reinstall?

dreamsource-tai, 2025-02-18:
@codepox 
type "./clean" to clean the temp files.

dreamsource-tai, 2025-02-18:
@codepox 
Please install x11 and replace Wayland to try again.

codepox, 2025-02-18:
Most Linux distributions ships with Wayland now.
Also I need Wayland for other software.
Are you planning to test and support Wayland?

codepox, 2025-03-01:
That's a different problem, you are right sometimes a popup modal window opens on another screen, but in that case the entire UI is unresponsive until that window is closed.