DreamSourceLab/DSView #855: DSView-U2P20-数据记录仪模式录制卡住

simon-sino, 2025-02-11:
在许多时基下都出现卡死的问题，不在记录新数据，在1kHz@1.16d下暂未发现卡死。
[image]

dreamsource-tai, 2025-02-12:
@simon-sino 
时基大则采集慢，基基小则采集快。采集慢的时候看起来就像卡。
你说的卡具体是？

simon-sino, 2025-02-12:
> [@simon-sino](https://github.com/simon-sino) 时基大则采集慢，时基小则采集快。采集慢的时候看起来就像卡。 你说的卡具体是？

挂了2小时，只录制了这点点数据，后面不再录制更多数据。

dreamsource-tai, 2025-02-13:
@simon-sino 
其实没什么问题，你看你的采集时间是2day，这个很长的。记录仪就不像示波器那样控制时基，是控制的采集时间。

simon-sino, 2025-02-13:
[image]
挂了一个多小时，只录制了200秒不到，后面的3000多秒一点没录，怎么会没问题呢？停止采样后没有出现后面的数据。

simon-sino, 2025-02-13:
问题是：有时候只采样了很小的一段时间，就不在录制新数据了，比如挂了1小时只有最开始的200秒被录制了，后面就不录了。

dreamsource-tai, 2025-02-13:
@simon-sino 
可能是usb连接不稳定或者供电不足导致的。不要过hub，直接接主板自带的usb端口，如果是台式机，不要接前面板的USB端口。
请再试试。

simon-sino, 2025-02-14:
[image]一小时只录制了400秒。
使用笔记本的USB3口，供电足够的。
而且使用1kHz 1.16d模式采样就没这个问题，能滚动录制几天。

simon-sino, 2025-02-14:
很容易复现，建议实测一下。

simon-sino, 2025-02-17:
使用1kHz 1.16d模式采样，周末滚动录制2天没问题。

dreamsource-tai, 2025-02-18:
@simon-sino 
我人已复现了这个问题，非常感谢你的反馈。

simon-sino, 2025-07-02:
请问这个问题有能被解决的风险吗？