DreamSourceLab/DSView #867: SD card in SPI mode not listed in 1.3.2 but listed in 1.3.0

lmk6, 2025-04-03:
As the title says, the SD card in SPI mode is not listed as a decoder but is present in the decoders directory. Interestingly enough, the closed issue (https://github.com/DreamSourceLab/DSView/issues/768) has the faulty code in the release anyway, which I imagine is related. I tried fixing it manually but it's still not listed.

dreamsource-tai, 2025-04-07:
@lmk6 
Can you display a list of other decoders?
I can list several decoders by entering the keyword 'sd'.

lmk6, 2025-04-07:
Yes, everything works except SD in spi mode.

lmk6, 2025-04-07:
@dreamsource-tai 
Just to make it completely clear, this is a build version I installed after downloading from here: https://www.dreamsourcelab.com/download/
It's for windows, so I can't really build it from the main repo.

dreamsource-tai, 2025-04-08:
@lmk6 
Please clone the source code from github, and copy the decoders to the main directory of the program. Try again.

lmk6, 2025-04-08:
I can confirm it's a valid solution to this. Thank you.

closed 2025-04-08 by lmk6