DreamSourceLab/DSView #877: Keyboard shortcuts, please

geonicz, 2025-06-15:
Hello guys!
Thanks for your hardware(U3P100 and dsTouch series) and software.
I've got both scopes. The only one issue with U3P100 is absence of shortcut keys for application.
It's very inconvinient to use mouse working with different hardware, clicking 
- AUTO button
- +-V,
- +-Time
- Start,Stop capture / Single run
Those basic functions are very essential with tactile responce from keyboard.
As I know, now there's no keyboard shortcuts assigned to those actions(maybe I haven't found them on my MacOS).
Can you add support?

dreamsource-tai, 2025-06-16:
@geonicz 

[image]

dreamsource-tai, 2025-06-16:
@geonicz 
Sorry, I really can't see what hotkeys are on Mac, but they can be seen on Windows and Linux.