DreamSourceLab/DSView #878: Unable to move the window by click+drag

M3C-Owx, 2025-06-15:
DSView v1.3.2
openSUSE Tumbleweed
Kernel:  6.14.6-1
DE: Plasma 6.3.5
Display Server: Wayland

**Expected Behavior: Normally, when you click and drag the top bar of a window, the window should move across the screen.**
----
**Observed Behavior: In my case, clicking and dragging the top bar does not result in any movement of the window. The window remains stationary regardless of the drag action.
Additionally clicking on top bar and moving mouse upwards in some cases resizes the window vertically**

**Everytime when I click on top bar and move mouse - a message in terminal comes up**:
> DSView: The top is out of range.

----

**Workaround: Alternatively, I can move the window by right-clicking the app on the taskbar, selecting 'More,' and then choosing 'Move' to reposition it using the arrow keys or moving mouse.**
----

**Additional troubleshooting steps performed: Tried Flatpak version, official openSUSE Repo, and tried to compile the app myself (QT6)**

-----
Debug-log:
`Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
Qt depends on a UTF-8 locale, and has switched to "C.UTF-8" instead.
If this causes problems, reconfigure your locale. See the locale(1) manual
for more information.

DSView: ----------------- version: 1.3.2-----------------
DSView: Qt:6.9.1
DSView: 2025-06-15 23:13:54
DSView: GetAppDataDir:"/usr/local/share/DSView"
DSView: GetFirmwareDir:"/usr/local/share/DSView/res"
DSView: GetUserDataDir:"/home/REDACTED/.local/share/DreamSourceLab/DSView"
DSView: GetDecodeScriptDir:"/usr/local/share/libsigrokdecode4DSL/decoders"
sr: lib_main: Init libsigrok.
sr: lib_main: Scan all connected hardware device.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found new DSLogic device count: 0
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
sr: backend: Register hotplug callback.
sr: lib_main: Hotplug thread start!
/usr/local/share/libsigrokdecode4DSL/decoders/arm_etmv3/pd.py:215: SyntaxWarning: invalid escape sequence '\s'
  instpat = re.compile('\s*([0-9a-fA-F]+):\t+([0-9a-fA-F ]+)\t+([a-zA-Z][^;]+)\s*;?.*')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_etmv3/pd.py:216: SyntaxWarning: invalid escape sequence '\.'
  branchpat = re.compile('(b|bl|b..|bl..|cbnz|cbz)(?:\.[wn])?\s+(?:r[0-9]+,\s*)?([0-9a-fA-F]+)')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_etmv3/pd.py:217: SyntaxWarning: invalid escape sequence '\s'
  filepat = re.compile('[^\s]+[/\\\\]([a-zA-Z0-9._-]+:[0-9]+)(?:\s.*)?')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_etmv3/pd.py:218: SyntaxWarning: invalid escape sequence '\s'
  funcpat = re.compile('[0-9a-fA-F]+\s*<([^>]+)>:.*')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_itm/pd.py:116: SyntaxWarning: invalid escape sequence '\s'
  instpat = re.compile('\s*([0-9a-fA-F]+):\t+([0-9a-fA-F ]+)\t+([a-zA-Z][^;]+)\s*;?.*')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_itm/pd.py:117: SyntaxWarning: invalid escape sequence '\s'
  filepat = re.compile('[^\s]+[/\\\\]([a-zA-Z0-9._-]+:[0-9]+)(?:\s.*)?')
/usr/local/share/libsigrokdecode4DSL/decoders/arm_itm/pd.py:118: SyntaxWarning: invalid escape sequence '\s'
  funcpat = re.compile('[0-9a-fA-F]+\s*<([^>]+)>:.*')
/usr/local/share/libsigrokdecode4DSL/decoders/ds1307/pd.py:51: SyntaxWarning: invalid escape sequence '\/'
  l += [('bit_' + re.sub('\/| ', '_', b).lower(), b + ' bit') for b in bits]
/usr/local/share/libsigrokdecode4DSL/decoders/ds1307/pd.py:55: SyntaxWarning: invalid escape sequence '\/'
  ['BIT_' + re.sub('\/| ', '_', b).upper() for b in bits] + \
DSView: Set default device.
sr: lib_main: Start activating device.
sr: lib_main: Activating device name: "Demo Device".
sr: lib_main: To open device, name:"Demo Device"
sr: lib_main: Activating device end.
DSView: Switch to device "Demo Device" done.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
DSView: Read region info, x:-6, y:0, w:1702, h:299, isMax:0
DSView: Last display:DP-3
DSView: Screen name:DP-3, region, left:0, top:0, width:2560, height:1440
DSView: Restore, screen:DP-3, x:0, y:0, w:2560, h:1440
DSView: Normal region, x:-6, y:0, w:1702, h:299
DSView: Init form, x:-6, y:0, w:1703, h:299, k:1
DSView: The top is out of range.`

dreamsource-tai, 2025-06-16:
@M3C-Owx 
Please use the qt5, and try agin. Thank you!

dreamsource-tai, 2025-06-17:
@M3C-Owx 
How many screens do you have? If there are two, they need to be arranged horizontally.
Also, what is your window system? I'm not sure if there are any issues with the Wayland system. If your current system is Wayland, please try switching to the X Window System.

acedogblast, 2025-08-15:
Run with `QT_QPA_PLATFORM=xcb`